zevero/avr_boot #4: ATmega2560 support

zevero, 2015-11-05:
We are looking for someone looking deeper into ATmega2560 support!

This is copied from @per1234 in Issue #1 

I managed to get the ATmega2560 to load a program from the SD card but it's far from ideal. I discovered the program will run under these circumstances:
- Set high fuse to DB instead of DA to disable the Boot Reset vector(BOOTRST=1).
  - The standard Arduino boards.txt enables the Boot Reset vector(BOOTRST=0) as does your hfuse setting in README.md.
- Use Atmel AVRISP mkII to Burn Bootloader instead of USPasp.
  - I normally use my USBasp due to the mkII not working correctly for Burn Bootloader with current versions of Arduino IDE for me without some ugly workarounds.
- Press the reset button after powering the board with the SD card attached.
  - The program has been loaded into the MCU flash but doesn't run until the reset button is pressed. With the other boards the program begins to run immediately after it has been loaded from the SD card.

After the program has been loaded from the SD no other programs will load from the SD card until the bootloader is burned again. If I change any of the above conditions it doesn't work. With all of these conditions it works consistently. I have no idea what this means, it's a bit more advanced than my current skill level.

JaspaJami, 2016-04-19:
Any news on this one?

zevero, 2016-04-19:
Any help in this issue is welcome.

JaspaJami, 2016-04-19:
Most likely im not able to help, but lets see when i have time to try it in 2560.

per1234, 2016-08-13:
I did more testing of avr_boot on ATmega2560:

Programmer(firmware) | 6.0.1/DA | 6.0.1/DB | 6.3.0/DA | 6.3.0/DB
---|---|---|---|---
AVRISP mkII, USBasp(1.06, 1.05, 1.04) | no | yes | no | yes
Atmel USBasp(Baite) | no | no | no | no
AVRISP mkII | no | yes | no | yes
Arduino as ISP | no | no | no | yes

**Defininitions:**
- **6.0.1**: avrdude 6.0.1-arduino5(included with Arduino AVR Boards 1.6.11 and previous)
- **6.3.0**: avrdude 6.3.0-arduino2(included with Arduino AVR Boards 1.6.12)
- **DA**: `high_fuses=0xDA`
- **DB**: `high_fuses=0xDB`
- **USBasp 1.06**: [PeterVH/bperrybap's v1.06-alpha-2016-05-18](https://github.com/bperrybap/usbasp/tree/1.06-alpha/bin/firmware/usbasp-v1.06-alpha-2016-05-18) USBasp firmware.
- **USBasp 1.05**: [PeterVH's 1.05 ](https://github.com/PeterVH/usbasp/blob/master/bin/firmware/usbasp-v1.5.hex) USBasp firmware.
- **USBasp 1.04**: [fischl's v1.04](http://www.fischl.de/usbasp/usbasp.2011-05-28.tar.gz) USBasp firmware.
- **USBasp Baite**: Stock firmware on the Baite USBasp clones
- **no**: Program never runs.
- **yes**: Burn Bootloader, start with SD card, press reset, program runs, no other programs will load from SD card until you burn bootloader again.

I also tried the "with_bootloader" hex file generated by **Sketch > Export Compiled Binary** with the same results.

My suspicion is that with the `high_fuses=0xDB` seeting the bootloader gets overwritten by the program from the SD card and this is why no further programs can be loaded from the SD card. I have seen somewhat similar things happen with ATmega2560 using Arduino as ISP+avrdude 6.0.1-arduino5 or USBasp firmwares other than 1.05 and 1.06 on avrdude 6.0.1-arduino5 and 6.3.0-arduino2. If you do an Upload Using Programmer with either of those programmers on an ATmega2560 with the standard `high_fuses` value of 0xD8 the program never runs. If you change the `high_fuses` value to 0xD9 then the program does run after Upload Using Programmer but if you Burn Bootloader and then do an upload over serial the bootloader is overwritten and you can't do any more serial uploads until you Burn Bootloader again. 
More information on both these issues at:
https://github.com/arduino/Arduino/issues/388.
Details on the USBasp issue at:
https://petervanhoyweghen.wordpress.com/2015/12/02/the-usbasp-and-atmega2560-mystery. 
Details on the Arduino as ISP issue at:
https://petervanhoyweghen.wordpress.com/2015/07/21/arduinoisp-reliability-and-portability-improvements.

hassin23ayz, 2017-09-09:
hello .. 

i have succeeded with atmega2560 

before calling the 0x0000 . you need to clear the EIND register for the extended memory addressing 

EIND = 0; 

this was suggested here https://www.mikrocontroller.net/articles/AVR_Bootloader_in_C_-_eine_einfache_Anleitung

kawasakizx10rr, 2018-08-24:
please may you share the working version for a Mega2560 hassin23ayz, I beg you.

shiryaevartyom, 2018-09-13:
> before calling the 0x0000 . you need to clear the EIND register for the extended memory addressing
> EIND = 0;

Please tell me more how to do this?
I'm watching the link
[AVR_Bootloader_in_C_-_eine_einfache_Anleitung Schritt_3](https://www.mikrocontroller.net/articles/AVR_Bootloader_in_C_-_eine_einfache_Anleitung#Schritt_3_-_Programmieren_des_Bootloaders)
But I do not understand where to insert **EIND = 0;**
do I need to write / add to a file? which file?
Please, give the file of the bootloader for upload in 2560

```
avrdude -c avrisp2 -p m2560 -P usb -e -u -F -U lock:w:0x3f:m -U efuse:w:0x05:m -U hfuse:w:0xDE:m -U lfuse:w:0xFF:m
avrdude  -c avrisp2 -p 2560 -P usb -b 115200 -U flash:w:bootloader_for_2560.hex -U lock:w:0x2f:m
```

ferhate, 2018-10-19:
You must change "main.c" file.

```diff
diff -r master/main.c atmega2560/main.c
247c247,250
< 		if (pgm_read_word(0) != 0xFFFF) ((void(*)(void))0)();	  //EXIT BOOTLOADER
---
> 		if (pgm_read_word(0) != 0xFFFF) {
> 			EIND = 0;
> 			((void(*)(void))0)();	//EXIT BOOTLOADER
> 		}
```

If you need the compiled hex file, please let me know.

litlbigchickn, 2018-10-26:
Hi Ferhat

I have been fiddling with the SD bootloader for 2560 since yesterday. I may just have gotten it to work a few moments ago. But if you could put up your (assuming you have AS7) version AS7 project with it - I would appreciate and reciprocate when and if my version gets to work - I am sanitizing mine removing the LED and Serial debug.

For some reason I had great difficulty getting the assembly file accepted in the project.

ferhate, 2018-11-19:
I've forked the repository and created a branch for ATmega2560 support.
https://github.com/ferhate/avr_boot/tree/atmega2560

@litlbigchickn 
For a working version for ATmega2560, please see the following commit: https://github.com/ferhate/avr_boot/commit/f97d4966b275a9ef84438976cf2b7755146b7615

For AS7 project files, please see the following commit: https://github.com/ferhate/avr_boot/commit/843614397c36843352ba059b83e16b85e2893c9c

@tech-shubham 
For compiled HEX files, please see build directory:
https://github.com/ferhate/avr_boot/blob/atmega2560/build/

alisyd, 2023-10-06:
> I've forked the repository and created a branch for ATmega2560 support. https://github.com/ferhate/avr_boot/tree/atmega2560
> 
> @litlbigchickn For a working version for ATmega2560, please see the following commit: [ferhate@f97d496](https://github.com/ferhate/avr_boot/commit/f97d4966b275a9ef84438976cf2b7755146b7615)
> 
> For AS7 project files, please see the following commit: [ferhate@8436143](https://github.com/ferhate/avr_boot/commit/843614397c36843352ba059b83e16b85e2893c9c)
> 
> @tech-shubham For compiled HEX files, please see build directory: https://github.com/ferhate/avr_boot/tree/atmega2560/build

Hello there! I am able to burn the bootloader but I cannot get it to boot from the SD card, I have put FIRMWARE.BIN on the SD card and I would want to know if there is something else I should be doing.

ferhate, 2023-10-14:
> Hello there! I am able to burn the bootloader but I cannot get it to boot from the SD card, I have put FIRMWARE.BIN on the SD card and I would want to know if there is something else I should be [...]

You should check the readme file.
https://github.com/zevero/avr_boot#put-your-sketch-on-sd-card

The common problem is the file system of SD card. The bootloader only supports FAT16/FAT32.

litlbigchickn, 2023-10-14:
Hi

I have gotten everything to work just fine. I have made a packet firmware
update system for nrf905 using the information from the bootloader you made
(not much left :-))

Unfortunately it is too long ago to remember what was the problem.

I made it work for arduino and AS7.

Yours sincerely
Nicolai Mathiasen

alisyd, 2023-10-14:
> You should check the readme file. https://github.com/zevero/avr_boot#put-your-sketch-on-sd-card
>
> The common problem is the file system of SD card. The bootloader only supports FAT16/FAT32.

Thanks! it works, but do you think is there a way I can trigger a software reset,  jump to bootloader from software, enabling the watchdog timer does not trigger boot from sd. Any help is much appreciated

shehabhassan, 2025-04-29:
i have issue with updating the code : 
https://github.com/zevero/avr_boot/issues/39