zevero/avr_boot #7: Make file not compiling with WinAVR 20100110

SergeantSeven, 2015-12-21:
Just changed a couple things in the make file. USE_LED set to 1, CS pin changed, all typical. but when doing 'make' in command line it gets to the .text part and comes up with an error 1. 
here is the output
'rm -rf *.o avr_boot.elf *.eps *.bak *.a *.bin
rm -rf pff/src/*.o uart/*.o
rm -rf *.lst *.map
rm -rf avr_boot.hex
avr-gcc -gdwarf-2 -Wall -Os -mcall-prologues -ffunction-sections -fdata-sections
 -mmcu=atmega1284p  -std=c99 -DBOOT_ADR=0x1F000  -DF_CPU=16000000   -DUSE_LED=1
 -DUSE_UART=0  -DCS_PIN=4  -DVARIANT_1284P=0    -c -o main.o main.c
avr-gcc -gdwarf-2 -Wall -Os -mcall-prologues -ffunction-sections -fdata-sections
 -mmcu=atmega1284p  -std=c99 -DBOOT_ADR=0x1F000  -DF_CPU=16000000   -DUSE_LED=1
 -DUSE_UART=0  -DCS_PIN=4  -DVARIANT_1284P=0    -c -o pff/src/pff.o pff/src/pff.
c
avr-gcc -gdwarf-2 -Wall -Os -mcall-prologues -ffunction-sections -fdata-sections
 -mmcu=atmega1284p  -std=c99 -DBOOT_ADR=0x1F000  -DF_CPU=16000000   -DUSE_LED=1
 -DUSE_UART=0  -DCS_PIN=4  -DVARIANT_1284P=0    -c -o diskio.o diskio.c
avr-gcc -c -mmcu=atmega1284p  -I. -x assembler-with-cpp -Wa,-adhlns=asmfunc.lst,
-gstabs -DBOOT_ADR=0x1F000  -DF_CPU=16000000   -DUSE_LED=1  -DUSE_UART=0  -DCS_P
IN=4  -DVARIANT_1284P=0  asmfunc.S -o asmfunc.o
avr-gcc -gdwarf-2 -Wall -Os -mcall-prologues -ffunction-sections -fdata-sections
 -mmcu=atmega1284p  -std=c99 -DBOOT_ADR=0x1F000  -DF_CPU=16000000   -DUSE_LED=1
 -DUSE_UART=0  -DCS_PIN=4  -DVARIANT_1284P=0  -Wl,-Map,avr_boot.map -Wl,--gc-sec
tions -Wl,--section-start,.text=0x1F000  -o avr_boot.elf main.o pff/src/pff.o di
skio.o asmfunc.o
c:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe: address
0x2013c of avr_boot.elf section .text is not within region text
c:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe: address
0x2013c of avr_boot.elf section .text is not within region text
make: *** [avr_boot.elf] Error 1'

Any ideas why?

SergeantSeven, 2015-12-21:
is the make file not up to par with current make file template standards?

per1234, 2015-12-21:
Well I wouldn't consider 2010 to be current. I believe that error indicates that the compile size is too large for the bootloader section. I get the same results trying to build with WinAVR but if I use the tools included with the Arduino IDE installation and then only use WinAVR for the tools no longer included with Arduino such as make.exe then it builds fine with the same makefile settings you're using(although I was getting larger compile sizes than zevero). Try this(works for me with Arduino IDE 1.6.5 - 1.6.7 and should probably work with other versions):
```
path C:\Program Files (x86)\arduino-1.6.7\hardware\tools\avr\bin;C:\winavr-20100110\utils\bin
make
```
You might need to edit the path to match your Arduino IDE install location.

SergeantSeven, 2015-12-21:
its the most current on Sourceforge... but I am new to this side of things. I wonder why there is a size difference compared to how zevero compiles on linux. it really shouldn't be different, 
I'll try your method.

SergeantSeven, 2015-12-21:
Though these are relating to older versions of the toolchain, I believe this answers all our issues and why Linux code is smaller then winavr code. I haven't tried the workaround posted in this thread yet but i feel it will help.
http://www.avrfreaks.net/forum/winavr-20070122-code-size-larger-20060421

SergeantSeven, 2015-12-21:
Closing issue, I downloaded Ubuntu and created a virtual machine for development use and was able to compile a 4040 byte .hex bootloader file directly on my removable drive. seems there is major issues with using WinAVR and the current source codes. lots of extra binary is being added for some reason, probably related to the post i linked to above. Don't know if anyone wants to tackle a workaround for it so windows users can continue to use the Winavr toolchain rather then going through the hassle of setting up a linux virtual machine.

closed 2015-12-21 by SergeantSeven

per1234, 2015-12-21:
Strange, the size of mine is 4014. Did you use the latest version of avr_boot and  these settings in the makefile:
```
MCU_TARGET    = atmega1284p # Target device to be used (32K or larger)
BOOT_ADR      = 0x1F000 # Boot loader start address [byte] NOT [word] as in http://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega1284p
F_CPU         = 16000000  # CPU clock frequency [Hz] NOT critical: it just should be higher than the actual Hz 
CS_PIN        = 4 # Arduino pin connected to SD CS. Supported values: 4, 8, 10, 53.
VARIANT_1284P = 0 # ATmega1284P variant: 0=avr_developers/standard, 1=bobuino, 2=sleepingbeauty
USE_LED       = 1 # Debug with two (defined in asmfunc.S)
USE_UART      = 0 # Debug on Serial. 0 ... deactivate or divider of http://wormfood.net/avrbaudcalc.php for baud rate!
```

SergeantSeven, 2015-12-21:
Due to the pinout of my project I modified the port mapping so the CS pin is pin 21 or PC5 and the LED pins are 1 and 3 PB1 PB3. but other then that everything is the same.

zevero, 2015-12-21:
I just happend to compile avr_boot for my project. Since you seem to be interested I put the compiled binary sizes into the readme. https://github.com/zevero/avr_boot
Note, that it has increased slightly since I added more UART debug info and some code to disable the watchdog reset.

per1234, 2015-12-21:
Thanks! Here's mine using the latest version of avr_boot:
3654/3704
3964/4014
3974/4034
20 bytes less on all but the last one which was 18 bytes less.

I tried the configuration listed in my last comment except with the boot section size increased to 4096 words using only WinAVR and it compiled to 4426 bytes so the cause of the original issue was the compile size being larger than the bootloader section size when building with WinAVR. Still I don't consider this a major obstacle for Windows users as they can always use the tools included with the Arduino IDE and the precompiled bootloader files in the avr_boot Boards package already cover all the most common configurations.

SergeantSeven, 2015-12-22:
still curious as to why my compile size was so different. I agree that the existing boards package covers the most common configs.