zevero/avr_boot #8: Bootlooping after Watchdog reset

JadinAndrews, 2016-07-27:
I am trying to reset the micro with a WDT in order to load the new code into memory, without physically having to push reset. The WDT reset method works fine on the optiboot bootloader, but gets wildly stuck with avr_boot. (LED flashes very fast) I installed the 328p bootloader through the IDE, so I'm not sure if that build includes the WDT fix or not.

I'll build it from source and see if that helps.

zevero, 2016-07-27:
You may have seen in the last commit of December 19th I included a fix for the WDT. I do not know if this is already in the binaries. Building from source should help.

JadinAndrews, 2016-07-27:
Great, compiling from source solved this, so the binary just needs to be updated. Funny though, I had to disable LED and UART in order for it to be > 4096 bytes. Maybe this is because of the extra WDT stuff?

zevero, 2016-07-27:
Please see my last commit ... I updated all the sizes after the WDT fix ... should only be 24 bytes more ... but this is compiled under linux. There was an issue with windows, or old compiler producing larger code ... but I do not remember well ... maybe you can find it in the other issues ...

per1234, 2016-07-27:
@zevero I'm happy to update the gh-pages branch with new binaries and Arduino IDE installation support(the current binaries in that branch are from v1.1.0, before this bug was fixed). Just create a release when you're ready. Glad to hear that watchdog bug was fixed. I haven't encountered it with avr_boot but I have with other bootloaders and it's very annoying.

zevero, 2016-07-27:
Oh yes I already fixed it last December after I run into that issue.
I made the release for v1.2.0 so you are ready to go.
Thank you!

JadinAndrews, 2016-07-27:
Well I'm glad it's already been fixed then, I compiled under arch, so I'd say a fairly new compiler was used, not sure why my sizes were off, I'll look into that later. For now I'm glad I got wdt reset working!
Thanks for the fast replies as well.

per1234, 2016-07-29:
Hi, I just wanted to give a status update on update of the gh-pages branch. The recent release of Arduino IDE 1.6.10 caused a serious issue where anyone installing avr_boot or almost any other AVR based 3rd party boards package will break the Arduino AVR Boards core. Adding a new release to the JSON file without fixing this issue will make that more likely to occur for the current users of avr_boot. I have come up with 3 different fixes but each has its own disadvantages. This issue has been reported to the Arduino developers and they say they will do a bug fix release soon to solve the issue. So I've been waiting to see what the best course is before proceeding with the update.

per1234, 2016-08-25:
@JadinAndrews or @zevero this issue can be closed.

closed 2016-08-25 by zevero