zevero/avr_boot #23: Set low a pin before bootloading

Emaxoso, 2017-03-17:
Hi, on my board to enable the SD card, i need to set low the digital pin 4 (PD4)
there is a way to do this before the bootloader try to access the sd?
Second question, can you provide the configuration of the Makefile for the atmega328p?

**Thanks man, i really appreciate your work** 👍

zevero, 2017-03-17:
Maybe it is easier if you just use https://github.com/zevero/avr_boot/tree/gh-pages
You will find compiled hex files here https://github.com/zevero/avr_boot/tree/gh-pages/avr/bootloaders
In Makefile you only have to change:
MCU_TARGET    = atmega328p
BOOT_ADR = 0x7000
(you need 2k which is at word 0x3800 which is (multiply by 2) byte 0x7000 
http://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega328p&LOW=42&HIGH=9B&EXTENDED=FF&LOCKBIT=FF
CS is the same pin 4 on PD4 ... it should be set to low by the boot code ...

Emaxoso, 2017-03-18:
Thanks for the help!
But my CS pin is 10 (PB2), the pin 4 (PD4) enable the current on the SD card
By the way i'm using a [Seeeduino Stalker ](http://wiki.seeedstudio.com/wiki/Seeeduino-Stalker_v3)

per1234, 2017-03-18:
[README.md](https://github.com/zevero/avr_boot#manual-installation) and the makefile both clearly document how to set the SD CS to any pin you like. If you want a much more simple solution just use the Arduino hardware package, which allows you to just select pin 10 from the **Tools > SD CS Pin** menu and then do **Tools > Burn Bootloader**. It doesn't get any easier! Installation instructions and other documentation for the hardware package are found here:
https://github.com/zevero/avr_boot/blob/gh-pages/README.md

Emaxoso, 2017-03-19:
@per1234  thanks for the help but maybe you didn't understand my problem my CS pin is the pin 10 (PB2) i have another pin for turning on the SD card, the pin 4 (PB4), the bootloader works flawlessly when i set the CS pin on 10 and the pin 4 (PB4) low manually, but i don't want to set this pin manually i hope i can add some code to the bootloader to set it automatically
- Thanks for help anyway 👍

per1234, 2017-03-19:
I understand now.

The decision of whether to add that feature is up to zevero. It seems like a fairly rare use case since that's not a very common board and the feature can only be enabled on that board by soldering the jumper pads. Maybe you could argue the case by researching whether other boards or shields have this feature and stating whether you would be willing to submit a pull request for the required changes.

It would probably be configured in the makefile something like:
```makefile
SD_EN = true  # Set the defined SD_EN pin low to enable SD card
SD_EN_PORT    = PORTB # Data Register of the SD enable pin
SD_EN_DDR     = DDRB # Data Direction Register of the SD enable pin
SD_EN_BIT = 4 # Bit of the SD enable pin
```
I am willing to do the work on the Arduino hardware package to make this feature available via a custom Tools menu in the Arduino IDE if the changes end up being made to the bootloader source and that is considered a desirable feature.

zevero, 2017-03-19:
Ok ... i presumed, that with EN you meant CS. Ok your really mean EN.

If you send me a pull-request with a configuration setup as proposed by @per1234 I am happy to add it. Please comment that SD_EN is only needed on boards, where EN is not hardwired, so that other users don't get confused.
Please check if you need some waiting time between pulling down and being able to read from SD.

Emaxoso, 2017-03-19:
What about something like this?
```
SD_EN		= true  	# Setup the enable pin (Needed only on board where is hardwired)
SD_EN_STATE	= LOW		# Set the state of the enable pin
SD_EN_PORT	= PORTB 	# Data Register of the SD enable pin
SD_EN_DDR	= DDRB 		# Data Direction Register of the SD enable pin
SD_EN_BIT	= 4 		# Bit of the SD enable pin
```
I added also the configuration to set the enable pin to high or to low, because i saw board where the enable pin goes high.
Can i pull request this configuration?
About the code i don't know how to implement this thing, i have no experience in programming bootloader for aduino, so i can only help with the documentation & configuration. 
I'm using this bootloader for a school project 👍 

- Thanks!

zevero, 2017-03-19:
I will not have time to setup, program and test ... but I believe you can do it easily once you look into init_leds(), led_power_on(), etc in main.c and asmfunc.S
led_power and led_write are just two pins which are set and unset (hardcoded in spi_pins.h :( )

After looking again into the code I think it may not really be worth the effort to configure everything so verbosely. We could just have SD_EN = true/false and set everything else in spi_pins.h - as done with the other pins...

SharkSharp, 2017-09-23:
Hi, I'm using arduino pro mini ATmega 328/P, but port 4 stays HIGH after boot. I am using the bootloader obtained by arduino's IDE. Is this the expected behavior?