zevero/avr_boot #27: SD Bootloader

Jono72, 2018-02-26:
Hi zevero,

I have been working on a project and would like to be able to flash my 1284P from SD card for updates, I have loaded your avr_boot, firstly i am unable to select standard pinout as it will not compile, so i left it set as avr-developers.com, this must be different from standard as it my program although runs and loads does not run as it should (i was using the mightycore bootloader before), if i reload that, the code runs fine.
Using avr_boot, I created BIN file as instructed and tried to load it , i moved my SD card CS to pin 10 (as i thought you should not use pin 4 on the 1284P the SS pin as a CS)
However the hex file does not load, it just boots up as before with program not operating as it should, when i reload Mighty Core (standard pin), my project runs fine. am i doing something wrong or is there another library to use? i tried downloading the Mighty 1284P standard as in your option selection, i just get a boards.txt error.

here is a copy of the error if i select standard Mighty1284P

panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0xc pc=0x4a75b7]

goroutine 1 [running]:
panic(0x54ef00, 0x12080030)
	/home/jenkins/go/src/runtime/panic.go:500 +0x331
arduino.cc/builder.(*SetupBuildProperties).Run(0x614ef0, 0x120ce000, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/setup_build_properties.go:84 +0xfa7
arduino.cc/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x614ef0, 0x120ce000, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_setup.go:59 +0x2b6
arduino.cc/builder.runCommands(0x120ce000, 0x12089dc0, 0x3, 0x3, 0x1, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 +0xe2
arduino.cc/builder.(*ParseHardwareAndDumpBuildProperties).Run(0x12089dec, 0x120ce000, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:180 +0xdc
arduino.cc/builder.RunParseHardwareAndDumpBuildProperties(0x120ce000, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:227 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:329 +0xf78
C:\Program Files (x86)\Arduino\arduino-builder returned 2

Error compiling for board ATmega1284/P.

Kind Regards

Jono

JadinAndrews, 2018-02-26:
You could try flashing my build for the 1284p?
https://github.com/JadinAndrews/avr_boot/blob/master/build/atmega1284p_cs4_16000000L_serial_fat32.hex

I managed to get serial working, the only issue is that the device signature is wrong and your flasher will think it is an ATmega 1284 non 'p'.

per1234, 2018-02-26:
>i am unable to select standard pinout as it will not compile, 

By default the ATmega1284P standard variant requires Mighty 1284P to be installed:
https://github.com/JChristensen/mighty-1284p/tree/v1.6.3
That is the cause of the error you got.

You can easily modify avr_boot to reference the MightyCore variant by simply changing line 384 of boards.txt from:
```
avr_boot_atmega1284.menu.pinout.standard.build.variant=mighty-1284p:standard
```
to:
```
avr_boot_atmega1284.menu.pinout.standard.build.variant=MightyCore:standard
```
Looking at the documentation I wrote now I see that I didn't explain that.

Although MightyCore is superior in every other aspect, I chose Mighty 1284P by default because it has the Sleeping Beauty variant, which MightyCore does not. Since that time I got the author of the Sleeping Beauty to provide an open source license so I am now able to add native support for that variant to avr_boot, which I planned to do for the next release. However, there has not been any development work merged to the avr_boot bootloader code since the last release so no reason for a new one.

>i moved my SD card CS to pin 10 (as i thought you should not use pin 4 on the 1284P the SS pin as a CS)

After you change the **Tools > SD CS Pin** option you need to to a **Tools > Burn Bootloader** again. Also, make sure you have the correct pin on the ATmega1284P connected. For the avr_developers variant Arduino pin 10 is mapped to PD2.

Jono72, 2018-02-26:
Awesome, thanks guys for the support, will give that a try :)

per1234, 2018-02-26:
>Im wondering if this could be a fuse setting or lock bit?

The correct fuses should be set when you do Tools > Burn Bootloader. I did fairly comprehensive testing with ATmega1284P before the release but there are a lot of possible combinations of settings so it's possible I didn't test your exact configuration and there's a problem. I'll try to get around to doing a test soon.

>i have added an ICSP port onto my board but this my USBASP programmer does not find the chip

ICSP uses SPI, the SD reader uses SPI also. So that statement makes me suspicious. However, if I understand correctly, you've already verified that the SD module is working correctly.

Jono72, 2018-03-01:
Yes everything working OK, ICSP is now working (I had the MISO and CLK pinout incorrect) as does the SD when running, ive had a look a fuse calculator online, I now understand how those are set but Im not sure what all the options provide, it just doesn't seem to access the SD and load the BIN file on boot as it should, is there a fuse setting that allows SPI to be initiated on boot?

Regards

John

Jono72, 2018-03-13:
Hi Guys, I know you are all very busy but if anyone cane me with this I would be most grateful, its driving me nuts.
So my Atmel 1284P is working fine, pinouts, SD card, ICSP programming socket all work fine, yet I cannot for the life of me get this bootloader to work, I am using CS on pin 10, as I have checked and all the advice back seems to be not to use Pin 4 (SS), I don't belive that's a problem because the SD card runs fine in the application, if I use either Visual Studio or the Arduino IDE the Bootloader does not seem to install correctly, I decided to use BIT BURNER (as its the only one I could find that supports 1284P) to try and load the bootloader direct I used "atmega1284p_var2_cs10_16000000L.hex"

this is the message from AVRDUDE when I try to load it.

Writing | ################################################## | 100% 81.68s

avrdude.exe: 130662 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\ABC\Documents\atmega1284p_var2_cs10_16000000L.hex:
avrdude.exe: load data flash data from input file C:\Users\ABC\Documents\atmega1284p_var2_cs10_16000000L.hex:
avrdude.exe: input file C:\Users\ABC\Documents\atmega1284p_var2_cs10_16000000L.hex auto detected as Intel Hex
avrdude.exe: input file C:\Users\ABC\Documents\atmega1284p_var2_cs10_16000000L.hex contains 130662 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 67.21s

avrdude.exe: verifying ...
avrdude.exe: 130662 bytes of flash verified
avrdude.exe: reading input file ""
avrdude.exe: error opening : Invalid argument
avrdude.exe: input file  auto detected as invalid format
avrdude.exe: can't open input file : Invalid argument
avrdude.exe: read from file '' failed

avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

What am I doing wrong, I have followed the instruction, and checked and double checked fuse settings, I increased the bootloader size to 4096 in the fuse settings, yet still I cannot get this to load the new BIN file.

Any ideas anyone??

Regards

Jono

per1234, 2018-03-13:
>if I use either Visual Studio or the Arduino IDE the Bootloader does not seem to install correctly

Please provide more information on what you meant by "does not seem to install correctly".

>I decided to use BIT BURNER (as its the only one I could find that supports 1284P) to try and load the bootloader direct I used "atmega1284p_var2_cs10_16000000L.hex"
>
>this is the message from AVRDUDE when I try to load it.

Is that the message from when you're using BIT BURNER?

per1234, 2018-03-13:
We'll continue the conversation here:
http://forum.arduino.cc/index.php?topic=534935
and then come back to GitHub if it turns out the problem is caused by a bug.