zevero/avr_boot #28: Changing CS to Pin 14

ratbertDS, 2018-08-12:
Hi for my project I have the CS Line wired to Pin 14 for the SD Card.  I'm a bit of a newbie here to changing bootloaders.  Can some one give me a clue to exactly what needs changing to make this work.  The CPU is a 328p 3,3v Device. 
Thanks!

per1234, 2018-08-12:
By "Pin 14", do you mean the Arduino pin number (as you would use with `digitalWrite()`, etc.) or do you mean physical pin 14 on the ATmega328P chip?

If the latter, what is the IC package of your ATmega328P?

ratbertDS, 2018-08-12:
So I am using a 3,3v Pro Mini.   
And I am connecting the SD Card as follows:
CS ->  Arduino pin A0
MOSI -> Arduino D11
SCLK -> Arduino D13
MISO -> Arduino D12

per1234, 2018-08-12:
OK. That's the information I needed. avr_boot doesn't use Arduino pin numbers to define the CS pin. It uses the PORT/BIT number. So the first thing is to determine the pin mapping of Arduino pin 14 on the Pro Mini. From this handy chart:
[image]

we can see that Arduino pin 14 is PC0 (PORTC, BIT 0).

You can define the CS pin port and bit either by editing [lines 9-11 of the Makefile](https://github.com/zevero/avr_boot/blob/master/Makefile#L9-L11) or via the `SD_CS_PORT`, `SD_CS_DDR`, and `SD_CS_BIT` arguments to `make`. I prefer the latter. So the correct `make` command is this:
```
make MCU_TARGET=atmega328p BOOT_ADR=0x7000 F_CPU=8000000 SD_CS_PORT=PORTC SD_CS_DDR=DDRC SD_CS_BIT=0 USE_LED=0 USE_UART=0 ADDED_CFLAGS="-Wextra -Wno-strict-aliasing" TARGET=$(MCU_TARGET)_cs14_$(F_CPU)L
```

ratbertDS, 2018-08-12:
Wow!  Thank you so much..