zevero/avr_boot #32: Question: Is it possible to use Software Spi?

hmantz, 2019-10-08:
Hi all! I have a project with an Atmega328P and I am using the hardware SPI pins for a different purpose. Is it possible to use non-standard pins to flash a new bootloader from a SD card?

Kieran-Weaver, 2020-09-05:
Yes, to use software SPI or any other form of SPI, just rewrite these functions:
```C++
void init_spi (void);		/* Initialize SPI port (asmfunc.S) */
void xmit_spi (BYTE d);		/* Send a byte to the MMC (asmfunc.S) */
BYTE rcv_spi (void);		/* Send a 0xFF to the MMC and get the received byte (asmfunc.S) */
```
They can be written in C/C++ as well as in AVR assembly.

FergusonMG, 2021-08-02:
Hello, we have an Arduino Mega2560 available [here](https://www.mercari.com/us/item/m17664631034/?ref=search_results) that come preinstalled with a microSD flash installer while still supporting updates through SPI.